easyw/kicadStepUpMod #190: exporting arrays cause to object has no attribute 'ShapeColor'

shprotru, 2023-10-26:
Related problem [here](https://github.com/easyw/kicadStepUpMod/issues/90).
here is the sample: [ITR9608-F.FCStd.zip](https://github.com/easyw/kicadStepUpMod/files/13173872/ITR9608-F.FCStd.zip)
try to select body, legarray and press export button at StepUp workbench.

however everything working fine while I am export the model manually in step format, I see in KiCad a colors that I set in FreeCAD.

failure log:
```
22:40:49  Running the Python command 'ksuToolsExportModel' failed:
Traceback (most recent call last):
  File "/home/eax/.var/app/org.freecadweb.FreeCAD/data/FreeCAD/Mod/kicadStepUpMod/./kicadStepUpCMD.py", line 1286, in Activated
    routineScaleVRML()
  File "/home/eax/.var/app/org.freecadweb.FreeCAD/data/FreeCAD/Mod/kicadStepUpMod/./kicadStepUptools.py", line 7445, in routineScaleVRML
    lbl=go_export(fullFilePathName)
  File "/home/eax/.var/app/org.freecadweb.FreeCAD/data/FreeCAD/Mod/kicadStepUpMod/./kicadStepUptools.py", line 2706, in go_export
    export(objs, fPathName, 0.3937, lbl)
  File "/home/eax/.var/app/org.freecadweb.FreeCAD/data/FreeCAD/Mod/kicadStepUpMod/./kicadStepUptools.py", line 2320, in export
    color.append(FreeCADGui.ActiveDocument.getObject(obj.Name).ShapeColor)

'Gui.ViewProviderLink' object has no attribute 'ShapeColor'
```

before export I set color attributes for array object.

easyw, 2023-10-26:
@shprotru 
simply make a union of your parts... it will solve the issue
[ITR9608-F-union.zip](https://github.com/easyw/kicadStepUpMod/files/13179825/ITR9608-F-union.zip)

shprotru, 2023-10-26:
@easyw wow, It works! I am very new to FreeCAD, can I able to ask you to add hint for kicadStepUpMod if it's see an array on export, like: "see array usage, go to part worbench and do a Union to unite parts"?

easyw, 2023-10-26:
please have a look at 
https://github.com/easyw/kicadStepUpMod/blob/master/demo/kicadStepUp-cheat-sheet.pdf
you can find it at Demo Menu under kSU FreeCAD WB

closed 2023-10-27 by easyw