easyw/kicadStepUpMod #191: Random weirdness with SolderJumpers

cdntinker, 2023-10-26:
Something very strange keeps happening with SolderJumpers.
They sometimes seem to import as bizarre shapes.
This seems to happen randomly.  Doesn't seem entirely consistent. Some work, some don't.
Hard to describe, so I'll try to attach some images.

Linux (LMDE5)
FreeCAD (0.19)
kicadStepUp (10.20.7)
[image]
[image]
[image]

easyw, 2023-10-27:
@cdntinker 
probably this is related tp the jumper footprint content. Not all the pad shapes are supported ATM.
Do you have a sample board to check?

cdntinker, 2023-10-27:
Apparently, GitHub doesn't like kicad_pcb files...
So I zipped it.
This is the board file that created the 2nd & 3rd pics above.

[zz.kicad_pcb.zip](https://github.com/easyw/kicadStepUpMod/files/13191856/zz.kicad_pcb.zip)

luzpaz, 2023-11-02:
@cdntinker you're using an obsolete version of freecad. Current stable is 0.21.1

cdntinker, 2023-11-02:
@luzpaz 
Yes, I'm running the version that works with the current version of my OS.
One day, I'll take the time & find the energy to fully upgrade my dev machine from LMDE 5 to LMDE 6 & find out what versions of things work & break.

luzpaz, 2023-11-03:
@easyw can you reproduce with the attached file ?

easyw, 2023-11-13:
this is related to the net tie fp and FreeCAD... the net-tie-fp geometry goes into coplanar OCC issues
https://discourse.mcneel.com/t/coplanar-surfaces-boolean-vs-planar-for-union-difference-and-intersect/152291
Unfortunately not all the kicad pad type can be fully converted to a clean geometry.
The OP may consider to move to a different net tie footprint as suggested here:
https://github.com/easyw/kicadStepUpMod/issues/196#issuecomment-1807756868