easyw/kicadStepUpMod #192: Document where to set `KISYS3DMOD`

xobs, 2023-10-27:
Both the program and the cheat sheet say to set `KISYS3DMOD`, but they don't say *where* to set it. Both places say to go into "Preferences", but they don't say where to set paths, and I have yet to be able to find it.

As a result, I get the following error every time I open the file:

[image]

There is a screenshot of the "STEP" view, but nothing there specifies PATH:

[image]

[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Singapore (en_SG)
Installed mods: 
  * kicadStepUpMod 10.20.7
[/code]

There should be documentation on which panel to adjust, rather than just sending users to "Preferences"

easyw, 2023-10-27:
> There should be documentation on which panel to adjust, rather than just sending users to "Preferences"

the doc could be improved, but:
1) the user should read it (at page 1) after the phrase: "your KISYS3DMOD value into the FreeCAD StepUp preferences page, located in the preferences system of FreeCAD (Edit menu -> Preferences)."
you can see "Just click the green icon: [kSU icon]"
2) the user should spend some time in reading the docs ... at page 10) you can see an image of the main kSU preferences
Preferences Page: "Preferences Page for configuring main parameters"

I will be pleased to accept PR for improving the docs... source of the cheatsheet pdf here:
https://github.com/easyw/kicadStepUpMod/blob/master/demo/kicadStepUp-cheat-sheet.odg
source file in LibreOffice format

An other note, that is related to FreeCAD use:
to find (show) a workbench preference page, you first need to load the workbench in FreeCAD.

xobs, 2023-10-28:
>  to find (show) a workbench preference page, you first need to load the workbench in FreeCAD.

Thanks! It was the last bit that was missing from the documentation. I've updated the documentation to reflect this, since I really was scratching my head over it.

PR #193 (2023-11-02): Cheat sheet: Document workbench requirement
Update the cheat sheet section on preferences to note that the required Preferences section won't show up unless the user is in the correct Workbench.

This closes #192
closed 2023-11-02 by easyw