easyw/kicadStepUpMod #195: ksuToolsAddTracks fail

myamigo, 2023-11-10:
# Description

I have a fairly complicated PCB (8 layers, 805 components) that I am importing into FreeCAD using KiCadStepUp. The "Load Board" command works fine and finishes in about 1 min. When I move to "Add Tracks" I hit a fault after a little more than 1 hr...

```
10:13:27  zones done
10:16:29  running time: 3850.488sec
10:16:29  running time: 3853.867sec
10:16:29  Sketch001
10:16:29  Running the Python command 'ksuToolsAddTracks' failed:
Traceback (most recent call last):
  File "C:\Users\scott\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\.\kicadStepUpCMD.py", line 4352, in Activated
    add_toberemoved = tracks.addtracks()
  File "C:\Users\scott\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\.\tracks.py", line 647, in addtracks
    topPads_cut_Name, temp_tobedeleted = cut_out_tracks(pcb_sk,topPads,ftname_sfx)
  File "C:\Users\scott\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\.\tracks.py", line 238, in cut_out_tracks
    FreeCAD.ActiveDocument.getObject('Board_Geoms'+tname_sfx).addObject(extrude)

<no object bound to built-in method addObject>
```

The fault occurs while processing the top layer. 

I've attached the log after scrubbing it a little. Give how log it takes to get to the fault I'd like to have some suggestions for how to speed this up. 

# Steps to reproduce

1. "Load Board" then "Add Tracks" ... Sorry but I can't share the project.
2. Review the log: [KiCadStepUp Issue (clean).txt](https://github.com/easyw/kicadStepUpMod/files/13320842/KiCadStepUp.Issue.clean.txt)

# FreeCAD | kicadStepUpMod Version
```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Canada (en_CA)
Installed mods: 
  * kicadStepUpMod 10.21.1
```

# KiCad Version

```
Application: KiCad x64 on x64

Version: 7.0.8, release build

Libraries:
	wxWidgets 3.2.2
	FreeType 2.12.1
	HarfBuzz 6.0.0
	FontConfig 2.14.1

Platform: Windows 11 (build 22631), 64-bit edition, 64 bit, Little endian, wxMSW

	wxWidgets: 3.2.2 (wchar_t,wx containers)
	Boost: 1.81.0
	OCC: 7.7.1
	Curl: 7.88.1-DEV
	ngspice: 41
	Compiler: Visual C++ 1936 without C++ ABI

Build settings:
	KICAD_SPICE=ON

```

myamigo, 2023-11-10:
I've managed to reduce the time to fault to ~17min by deleting duplicate circuits. It looks to be faulting in the exact same way. I'll try deleting a bunch of free vias and other things to see if I can speed it up.

myamigo, 2023-11-10:
I'm down to 8 min now.

easyw, 2023-11-10:
@myamigo 
thanks for the detailed feedback.
I tried your board but I cannot reproduce your issue...
I get only this warning message:
`Part::Extrusion: Link(s) to object(s) 'Sketch' go out of the allowed scope 'Extrude57'. Instead, the linked object(s) reside within 'Board_c493'
`
here my FC info:
```
OS: Windows 10
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
```
tested also with the one from the installer:
```
OS: Windows 10 build 18363
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
```

[StepUp_Issue#195-castellated-kv7.zip](https://github.com/easyw/kicadStepUpMod/files/13323381/StepUp_Issue.195-castellated-kv7.zip)
I downloaded FC from here:
https://github.com/FreeCAD/FreeCAD/releases/download/0.21.1/FreeCAD_0.21.1-Windows-x86_64.7z

Your issue and FC error seem similar to this issue https://github.com/easyw/kicadStepUpMod/issues/179

As a side note, please consider to add Grid origin to your kicad_pcb file... then all your sketches in FC will be aligned to this origin.
[StepUp_Issue#195-castellated-wGridOrigin-kv7.zip](https://github.com/easyw/kicadStepUpMod/files/13323491/StepUp_Issue.195-castellated-wGridOrigin-kv7.zip)

myamigo, 2023-11-11:
Hmmm. What version of KiCad are you running? I prefer aux=drill origin so the PCB lines up with other mech models.

myamigo, 2023-11-11:
That error mssg with sketch out of scope seems to be relevant. Does the mssg go away when the board outline is extended past the part?

easyw, 2023-11-12:
> Hmmm. What version of KiCad are you running? I prefer aux=drill origin so the PCB lines up with other mech models.

you can choose aux=drill origin as you prefer, 
but then you have to set in FC kSU pref the same. 
Then FreeCAD and KiCAD will use the same reference for mech objects

[image]

myamigo, 2023-11-12:
Thanks for the tip!

I solved my "no object bound to built-in method addObject" problem at line #238 of tracks.py 

I changed that line from:
```
        FreeCAD.ActiveDocument.getObject('Board_Geoms'+tname_sfx).addObject(extrude)
```

to 
```
        try:
            FreeCAD.ActiveDocument.getObject('Board_Geoms'+tname_sfx).addObject(extrude)
        except:
            print('EXCEPTION: Board_Geoms'+tname_sfx+'.addObject(extrude)')
            pass
```

And it happily gets by. I don't know exactly what is going on under the hood but I expect the edge connector is getting split into two zones groups with one being associated with the board because it is inside, and the other is left as an orphan. This change seems to handle the orphan.

This got me by that issue and on to another one involving net-ties that I will create another issue for.

Thank you for your help and a great FC add-on.

easyw, 2023-11-12:
@myamigo 
please test the board with the new update  https://github.com/easyw/kicadStepUpMod/commit/0f8c4d500b9495c26ec185f28adb7d3344d28136 ... 
it should solve the issue and keep the cut for tracks.

myamigo, 2023-11-13:
Confirming that your update gets by the issue. I'm testing it with my larger PCB now and I'll result  from that.

myamigo, 2023-11-13:
The larger PCB import completed and it looks much better but now I am noticing problems. 

There is a large rectangular "pad" that is the size of the overall GND pour included with the topPads face but it is located at the design origin (0, 0) where the grid origin is now at (200, 100). 
[image]

If I `Draft | Downgrade` the face I can select and delete the rectangle. Then I can select all of the downgrade shapes and Upgrade them gain I have what I expect... almost!

I am noticing random instances of pads missing...
Here is a 12x12 BGA in KiCad...
[image]
 
... and after the FC_ SU import...
[image]
... that is mostly showing a 9x9 pattern.  

This show show 2 pads and a net-tie (highlighted) ...
[image]

The ISP_BOOTn pad and net-tie next to it are missing in FC as is the footprint below...
[image]

In another spot I see a track area that has not been completely constructed. ... 
In KiCad...
[image]

When imported...
[image]
Several pads are missing in this view and the highlighted rectangle does not have a outline like the other objects. I think I can work with the output now so I can export to STEP an load back into my CAD package for assembly fit and rendering.

I am willing to try anything you might suggest but I have to get back on track.

myamigo, 2023-11-13:
I may be seeing more issues with teh use of complex pads as I note in https://github.com/easyw/kicadStepUpMod/issues/196#issuecomment-1808898430

I'll report what I find.

easyw, 2023-11-13:
If you are looking for rendering, you may look for kicad to blender plugin.
To get assembly checking, missing some pads shouldn't be an issue at all.
The main purpose of kicadStepUp is the mechanical integration and tracks are added just for this scope.

myamigo, 2023-11-14:
I've not looked at Blender yet but I will. I also looking for high quality renders of our product with the PCB in exploded view so our pro CAD package is best for that. Anyway, I'm getting closer to a solution.

myamigo, 2023-11-15:
My issues have been addressed with updated FP's as per #196.

closed 2023-11-15 by myamigo