easyw/kicadStepUpMod #196: Net-tie handling

myamigo, 2023-11-12:
With my larger project I've experienced problems with the use of net-tie components. I use them to isolate seed circuits that I replicate with the ReplicateLayout plugin. The results in corrupted pads that have been confused by an incomplete zone used in the construction process. As before I isolated the problem by stripping stuff out until I found teh one thing that was giving me grief. That would be net-ties, obviously!

I build on my previous example and added a net-tie footprint to expose the issue. 

My net-tie consists complex pad shapes and a colnecting polygon...
[image]

The dimensions are fine for DRC checks and the resulting fill is perfect.

Here the highlighted net-tie connects 2 different signals on the edge connector and I've filled the rest of the F.Cu layer with a GND pour.

[image]

This is fine in KiCad and when rendered to gerbers but I note something different when I import into FreeCAD using StepUp...

This is the result...
[image]

Note that I used the "fix" from issue #195 here.

The tracks and zones look fine but there's something going on with the net-tie pads. 

Zooming in with the topPads_6c70_cut face selected shows...
[image]
    
What you see here cause me major grief with the larger project that I can't begin to explain. This will kick off the discussion though.

myamigo, 2023-11-13:
Here's the test project for this issue... [StepUp_Issue#196.zip](https://github.com/easyw/kicadStepUpMod/files/13329611/StepUp_Issue.196.zip)

The composite pads for the net-tie consist of a circular pad + a rectangle so the inner edge of the larger pad is vertical. When imported into FreeCAD, we should end up with a single segment but for this example it is incomplete.

I did another test where I duplicated the 1st net-tie and assigned different nets to its pads and it had a different issue when imported with SU. One net-tie rendered 100% correctly and the other 0% (no copper pad at all). The test project will allow simple scenarios, all which should work.

easyw, 2023-11-13:
hi @myamigo 
this is related to the net tie fp and FreeCAD... the net-tie-fp geometry goes into coplanar OCC issues
https://discourse.mcneel.com/t/coplanar-surfaces-boolean-vs-planar-for-union-difference-and-intersect/152291
Unfortunately not all the pad 

Could you please try to change the NTfp with this one:
[NetTie_0.2mm-mod.kicad_mod.zip](https://github.com/easyw/kicadStepUpMod/files/13332943/NetTie_0.2mm-mod.kicad_mod.zip)

and here the pcb with this NTfp:
[StepUp_Issue#196-net-tie-mod-kv7.zip](https://github.com/easyw/kicadStepUpMod/files/13332946/StepUp_Issue.196-net-tie-mod-kv7.zip)

The composite pads for this net-tie consist of a circular pad + a rectangle pad (not a custom pad) + a polygon overlapping the rectangles.
[image]

myamigo, 2023-11-13:
Thanks for the direction. I have confirmed that this works for my simple test case so I applied it to my library of ~50net-tie FP's (multiple widths over any internal or external layer). I'm going to try importing my main PCB with the updated NT's and using the fix from #195 to see if I get through without issues.

I am noting that the net-tie does not have solder mask over it but it should (unless it overlaps an exposed pad of course). 
[image]

That's a minor  point give that a more serious problem has been fixed.

Thanks again!

myamigo, 2023-11-13:
OK so I have another type of net-tie in my design that is probably causing problems based on what you've told me.

This one allows me to cut a trace to isolate a circuit and  then reconnect with a blob of solder...
[image]

This consists of a custom pad with a circular base located to edge of the pad so connecting tracks will not have clearance violations with the other pad. Then I've attached two polygons to complete the shape. Here is the anchor...
[image]

I've reworked it as a trapezoidal pad with a polygon connector and I am trying this out with StepUp import.

easyw, 2023-11-13:
why not using a fp similar to the one I made and post, adapting simply a different size and spacing...

myamigo, 2023-11-14:
> why not using a fp similar to the one I made and post, adapting simply a different size and spacing...

Yes that's pretty much what I did. All good here I believe.

myamigo, 2023-11-14:
I found a few more footprints with complex pads. I rooted them out with a global file search. I'll update these and advise. This might solve my remaining issues.

myamigo, 2023-11-15:
I've discovered an issue with the  modified net-tie (NT)  footprint (FP)  when a via is located at one end of the NT.

I updated my test project ([StepUp_Issue#196.zip_V2.zip](https://github.com/easyw/kicadStepUpMod/files/13366837/StepUp_Issue.196.zip_V2.zip)) as follows...
[image]
Note the via and track layer change from p#4 - p#9.

With the original NT FP (NetTie_0.2mm_ORIG_CUSTOM) I get the following... 
[image]
The "pad" is broken into a number of parts that I believe are messing with FreeCad when the difference is completed.

With your version (NetTie_0.2mm-mod_easyw) I end up with... 
[image]

With my updated version based on what you provided (NetTie_0.2mm_NOT_CUSTOM) I similarly get...
[image]

It seems like FreeCad is having a problem with a via located at the center of a circular pad so I thought I would try something different by converting the circular pad to a slight oval with an "Offset shape from hole" setting to locate the pad anchor in the same location as the circle (NetTie_0.2mm_NOT_CUSTOM_OVAL)...
[image]
I also shifted the companion rectangular pads towards the  center of the FP by 0.1mm to match the larger oval pad shape...
[image]
Then I adjusted the size of the connecting polygon, although that isn't really necessary.

This works when imported with StepUp...
[image]

I applied this to my larger project and tested the FP to make sure I had no new DRC issues in KiCad.... All good with that!

Thinking the original issue was only associated with the vias/holes at the center of a custom pad, and that is a problem for FreeCad's difference function, I also reverted all other FP's that had custom pads changed to non-custom back to custom. 

Then I imported a portion of the larger design into FC with StepUp and that worked as well...
[image]

I'm  good on this now... and moving on to the next issue [image]

closed 2023-11-15 by myamigo