easyw/kicadStepUpMod #197: Copper, Mask & Silkscreen Positions

myamigo, 2023-11-13:
KiCad determines the PCB thickness using the mask and as well as copper and dielectric.  It considers the silkscreen to be 0mm thick, which is, of course, unrealistic.

For my test project this is the stack-up....
[image]

The thickness is 0.159mm (would be ~1.6mm for a typical 2-layer board but I cut this back from an 8-layer PCB). That's not important for this discussion.

When this is imported into FreeCAD with StepUp the PCB thickness is set to be 0.159mm so it includes the copper and mask. 
[image]

Therefore it makes sense to position the copper, mask and silkscreen faces as close as possible to the PCB top & bottom surfaces. I would suggest the following:
* F.Cu ... 1µm over front/top face
* F.Mask ... 2µm over front/top face
* F.Silkscreen ... 3µm over front/top face
* B.Cu ... 1µm below bottom face
* B.Mask ... 2µm below bottom face
* B.Silkscreen ... 3µm below bottom face

These setting avoid overlap interferences and results in a board that looks to be close to teh intended thickness. This is fairly minor but I've been focussing on detail lately so i thought I'd share my thoughts.

I also note that the topPads face is placed at 0.02mm while the tracks and zones are at 0.01mm. I'm sure the intention is to have them coplanar so I've pointed that out.

easyw, 2023-11-13:
the pcb board gets the full pcb height from kicad settings, to allow pcb + 3D models to be a watertight model (SMD only)
other layers must be detached to avoid z-fighting (flickering in the display of your 3d model)...
Please consider that silks and pads or tracks are just here for mechanical integration (to check if there are issue with box or anchors).
If you need a realistic pcb stackup you need to move to [fcad_pcb](https://github.com/realthunder/fcad_pcb), which is the library used to build up tracks.

myamigo, 2023-11-13:
Gotcha and thanks for the suggestion. fcad_pcb looks like it could be very useful if I need to do thermal analysis or if I need high detail. StepUp will do for what I need right now if I can sort out remaining issues, which seem to be related to my use of net-ties.

You can close this.

closed 2023-11-13 by easyw