easyw/kicadStepUpMod #198: RecursionError: maximum recursion depth exceeded

myamigo, 2023-11-13:
Building on improvements resulting from #195 & #196 the track import does get further long but then it fails in a new way.
[image]

Reviewing the component tree I see a recursive extrude...
[image]

I expanded it a number of times and it goes on and one so this is likely where the problem is occurring.

Here's the log for this... [FC_StepUp_Recursion_Error.txt](https://github.com/easyw/kicadStepUpMod/files/13342249/FC_StepUp_Recursion_Error.txt)

easyw, 2023-11-13:
you probably run out of memory... your tracks are too complex too be handled by this tool.
Test it with fcad_pcb or you could try with kicad 7.99 that has an option to export tracks to step... but I think it may be a big jpb to do.

myamigo, 2023-11-14:
Yeah could be my project is too complex. I had a RecursionError problem with a different Python tool that was resolved by increasing the recursion limit. From the output it looks like the limit is currently 1000. For the other tool I increased this to 5000 and that got me by the issue so I tried that as well here by adding `sys.setrecursionlimit(5000)` to  `kicadStepUptools.py` . Seems to have worked.

However, I am still having problems with an incomplete `topPads_a521_cut` object. It would be helpful if the original pads object that is built from the footprints when the "Add Tracks" command is run was saved in a hidden state in the project tree under Board_Geoms_a521. That way I could at least generate pads and a mask with the PCB outline using boolean operations. I wouldn't have tracks or zones but I am not getting that anyway. 

I was digging through the source looking for where that might be done and I found the spot in `kicadStepUpCMD.py` where I can block removal of the intermediate objects. Changing as follows...
[image]

... will leave the objects in place and i can see what I need there...
[image]

I was hoping to see an object name with the print() statement but that doesn't show what I need, which is to identify the `Common_Top71` object so it can be skipped.

I see that you provide the option to not import zones...
[image]
If you also provided that option for tracks then I might end up with a solution. Is that something you could implement?

myamigo, 2023-11-15:
The recursion error seems to have been resolved with the net-tie related changes. I know how to save intermediate objects that you build when adding tracks so I can enable that whenever I need. 

All good here! Thanks for you help and a truly great add-on!

myamigo, 2023-11-18:
@easyw the recursion error resurfaced after I tried importing my PCB with all pads, tracks and zones enabled. I then edited `kicadStepUptools.py` to enable the lines:
```
max_recursion_limit=5000  # kSU issue#198
sys.setrecursionlimit(max_recursion_limit)
```
 ... and the board was built without the recursion fault so this looks to be a good permanent change.

easyw, 2023-11-19:
closing w/
https://github.com/easyw/kicadStepUpMod/commit/938a7f564b8dc87da07fd593e412b0c3e2dc47e4

closed 2023-11-19 by easyw