easyw/kicadStepUpMod #199: wrong orientation of holes from footprint

MasteringTheMess, 2023-11-15:
I have a connector that adds two holes in the pcb for positioning.
When I import this board into FreeCad via Kicad stepup the orientation of the holes is wrong if I rotate the footprint:

[image]
[image]

It's fine in the KiCad 3D-Viewer though:
[image]

Heres the pcb-file:
[StepUp_test.zip](https://github.com/easyw/kicadStepUpMod/files/13363171/StepUp_test.zip)

easyw, 2023-11-17:
@MasteringTheMess 
thanks for the testing file...
it should be solved now
https://github.com/easyw/kicadStepUpMod/commit/4f207d1646bb49606f42997bfb7d126592804d00

closed 2023-11-18 by easyw

MasteringTheMess, 2023-11-20:
Can confirm it works now, Thanks!

Just for the record: the problem did not occur when NPTH were used instead of edge cutouts