easyw/kicadStepUpMod #200: Feature Request: Skip Importing of PCB Tracks

myamigo, 2023-11-15:
For large designs it would be nice to have the option to skip importing of tracks. If zones are also skipped using the existing `Skip importing pcb zones` settings flag, then `Add Tracks`  would only operate on the pads and run much quicker.
[image]

This would be fine in many cases where seeing where an unpopulated component are located on the PCB would be of interest. I'm thinking about presentation of product variants.

easyw, 2023-11-17:
@myamigo 
it should be available now
https://github.com/easyw/kicadStepUpMod/commit/4f207d1646bb49606f42997bfb7d126592804d00

myamigo, 2023-11-17:
Nice! Works well thank you.

I though vias might be bundled with tracks so if we exclude tracks we also exclude the vias but that's not the case. Cutting vias into the pads adds a lot of time and vias are added un-tented so they can create clutter, especially for a heavily stitched PCB (like mine).

Could vias be included/excluded with tracks? Normal thru-holes defined in footprints would be maintained with the corresponding pads.

easyw, 2023-11-18:
@myamigo 
if you need to esclude vias you can:
a) save a copy kicad_pcb without vias and import that 
b) use the parameter `via_bound=-1` in tracks.py at `pcb = kicad_parser.KicadFcad(filename,merge_pads=False)`
have a look at here:
https://github.com/realthunder/fcad_pcb/issues/22

myamigo, 2023-11-18:
Thanks for the suggestion. @easyw . I modified the statement as follows so vias are only included when tracks are include.

`pcb = kicad_parser.KicadFcad(filename,merge_pads=False, via_bound=(-1 if skip_import_tracks else 0)) `

This seems like a good general change,

easyw, 2023-11-19:
https://github.com/easyw/kicadStepUpMod/commit/938a7f564b8dc87da07fd593e412b0c3e2dc47e4

closed 2023-11-19 by easyw