easyw/kicadStepUpMod #202: Import blacklist stopped working?

fullmoon6661, 2023-11-22:
Our simple import blacklist `C_1206;C_0805;C_0603;C_0402;R_1206;R_0805;R_0603;R_0402;` stopped working after recent update, every component that should be caught by the filter still gets imported.

easyw, 2023-11-23:
@fullmoon6661 
it sshould be now restored https://github.com/easyw/kicadStepUpMod/commit/f67faa3dfe503cce9d4f0edede204d3db0e89471

fullmoon6661, 2023-11-23:
Thank you @easyw. It now gives `... black-listed module(s)` popup but filtered components still getting imported.

easyw, 2023-11-23:
@fullmoon6661 
could you please post a simple PCB with local 3d modules with the blacklist string to test?
Use KYPRJMOD for portable 3d models.

fullmoon6661, 2023-11-23:
Here is a [TESTPCB.zip](https://github.com/easyw/kicadStepUpMod/files/13453098/TESTPCB.zip) with official KiCad capacitors and resistors footprints from 1812 to 0201, and using blacklist `C_1206;C_0805;C_0603;C_0402;R_1206;R_0805;R_0603;R_0402;` it should not import anything except 1812.

I'm not sure what do you mean with "local 3d modules" or "KYPRJMOD for portable 3d models". I'm using vanilla KiCad 7.0.7, and FreeCAD 0.21.0 with latest KicadStepUp, all with default settings along with these path:
Within KiCad:
[image]
Within KiCadStepUp:
[image]

easyw, 2023-11-23:
@fullmoon6661 
to skip your models, you need to add the full 3d model name in blacklist.
In your case i.e. "R__0402” should be the corresponding full name of the 3D model which is "R__0402_1005Metric".
Fixing this in your blacklist string it should work correctly.

fullmoon6661, 2023-11-23:
Oh you're right, I changed with the full name now and it works perfectly. I must have mistaken my old blacklist string.

Also just to note giving a single item in the string without a ";" doesn't work, this was causing problem with my testing.

Thank you, and sorry for the trouble @easyw.

closed 2023-11-23 by fullmoon6661