easyw/kicadStepUpMod #203: Bottom side Cu and silkscreen are inverted when STEP model is imported to Fusion 360

ArnieO, 2023-12-13:
In general, the tool works just great for my use, but there is a small snag that I have to fix manually - but it would be great if it could be fixed directly in kicadStepUp:

In my workflow, I export STEP model from FreeCAD, and import the STEP model into Fusion360.

In F360 I then have to revert the face orientation on the PCB bottom side layers (Cu and silkscreen) to make them appear correctly.

It would be great if this could be fixed in kicadStepUp, so they have correct orientation in the STEP model!

easyw, 2024-02-01:
hi @ArnieO 
this seems an issue in fusion360.
Have you tried to open the step w i.e. cadassistant of open cascade?
https://www.opencascade.com/products/cad-assistant/

ArnieO, 2024-02-01:
Hi @easyw 
Thank you for your feedback and suggestion. I have installed CAD assistant now and I cannot see the problem there. However, it is difficult to be sure, as it depends on whether CAD assistant paints the surfaces with the same colour on both sides.

Here is the board in CAD assistant:
[image]

And the same STEP model imported to Fusion360, where the bottom layer tracks and zones are reversed:
[image]

If I hide board and top layer geometry in CAD assistant, I see this when looking from the top side:
[image]

And this when looking from the bottom side:
[image]

So it is not evident to me if I can here identify whether the layers are reversed or not. Any advise from your side on how I can find out?

ArnieO, 2024-02-01:
Seems to be rendered with same color on both sides:

[image]
[image]

easyw, 2024-02-01:
@ArnieO 
it seems we had this issue few time ago:
https://github.com/easyw/kicadStepUpMod/issues/25
there is a simple workaroud to fix fusion issue in giving the colors:

> It was possible to create a mirror of the offending piece, strangely enough this made the original piece appear as expected.
> After this the (new) mirrored piece can be removed. (You will see that the mirrored piece again has the color).
[image]

https://github.com/easyw/kicadStepUpMod/issues/25#issuecomment-667537219
Not sure if this is still a solution in latest fusion release.

ArnieO, 2024-02-02:
Thank you for your proposition. The workaround creates a new, version of the selected items, but does not change the original (as indicated in that comment). 

So I'll stick with my current workaround: Switch to Surface mode, select the surfaces that need to be reversed - and then reverse them.

I'm not convinced that the issue is F360 STEP import, as the view in cadassistant of open cascade renders both sides of the layer with same color - so not really helpful to verify the STEP model. But you seem to be sure the issue is with F360 STEP import?

easyw, 2024-02-02:
@ArnieO 
do you have a kicad_pcb file and freecad step file to test?

ArnieO, 2024-02-02:
Sure, please see attached ZIP file. 
[Testjigg Pow-P1_v2.0.zip](https://github.com/easyw/kicadStepUpMod/files/14143618/Testjigg.Pow-P1_v2.0.zip)

easyw, 2024-02-03:
@ArnieO 
I tried your files and in CADA everything is as expected... colors are preserved
IMO the issue is in F360 STEP import process.

ArnieO, 2024-02-03:
@easyw Thank you very much for your efforts!
That sounds very good, but just to be sure: Can you in CADA see difference on front and backside of a surface? In order to see the issue, you have to make PCB core and top layers invisible, and (in some way) verify that the front and back side of the bottom layers (tracks and zones) are not inverted.
Are you able to verify what is front and back side of the surfaces in CADA - or is that maybe a notion that does not exist in the STEP model? It could very well be as you says, the F360 import that tries to make solids, and makes a mistake on these surfaces.

easyw, 2024-02-04:
I tested the top tracks from top and bottom tracks from bottom and the colors were just correct.
I would consider this an issue in f360.

ArnieO, 2024-02-04:
Thank you very much for your kind assistance in sorting this out!

closed 2024-02-04 by ArnieO