easyw/kicadStepUpMod #204: Workbench failure: No module named "request"

transistorgrab, 2023-12-22:
Today I updated kicadStepUpMod and was greeted with an error message while opening the workbench.
[image]

Freecad version is 0.21.1
Plugin version is 10.22.4

easyw, 2023-12-23:
@transistorgrab 
could you please post your full FC info?
it is strange the python module 'requests' is not available...

seflm, 2023-12-26:
Got the same error

installed through built-in extension manager
Freecad version is 0.21.1
Plugin version is 10.22.5

`18:15:40  No module named 'requests'
18:15:40  Traceback (most recent call last):
  File "<string>", line 250, in Activated
  File "C:\Program Files\FreeCAD 0.21\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "C:\Users\marek\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\.\commits_num.py", line 1, in <module>
    import requests
  File "C:\Program Files\FreeCAD 0.21\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)`

transistorgrab, 2023-12-26:
@easyw: I installed FreeCAD from the installer.
Full Version Info:
```lua
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * Assembly4 0.50.6
  * dxf-library
  * kicadStepUpMod 10.22.4
```

Output from Python console:
```python
>>> import requests
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "D:\Program Files\FreeCAD 0.21\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
ModuleNotFoundError: No module named 'requests'
```

However, I found that this throws no error:
```python
from urllib import request
```

I hope this helps. :)

easyw, 2023-12-27:
It is strange there is no module on FC installer... I see if I can test on my PC.

easyw, 2023-12-27:
ok, I have downloaded FC021.1 from:
https://github.com/FreeCAD/FreeCAD/releases/download/0.21.1/FreeCAD-0.21.1-WIN-x64-installer-1.exe

[code]
OS: Windows 10
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
[/code]
and from the FC python console I can digit:
`import requests`
without errors...

easyw, 2023-12-28:
@transistorgrab 
could you please try the current update?
It should work Aldo without thr module 'requests'

transistorgrab, 2023-12-29:
```lua

OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * Assembly4 0.50.6
  * dxf-library
  * kicadStepUpMod 10.22.8
```

I just installed the current download and it says the same version as yours and the same as reported before.
I still see the same behaviour:

```python
>>> import requests
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "D:\Program Files\FreeCAD 0.21\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
ModuleNotFoundError: No module named 'requests'
```

I can confirm that I saw no problems using the Workbench for my use case. Just wanted to report the observation.
With the latest update of KSU the error report is already gone when opening the Workbench.

easyw, 2023-12-30:
@transistorgrab
thanx for your feedback.

closed 2023-12-30 by easyw