easyw/kicadStepUpMod #208: Can't load footprint due to character encoding mismatch.

uchan-nos, 2024-02-11:
I'm using KSU 11.1.6 on FreeCAD 0.21.2 and KiCad 7.0.10. OS is Windows 11.

When I tried to load a KiCad footprint by clicking "Load FootPrint" button, the following error occurred.

```
21:01:40  Running the Python command 'ksuToolsImportFootprint' failed:
Traceback (most recent call last):
  File "C:\Users\uchan\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\.\kicadStepUpCMD.py", line 5006, in Activated
    fps.addfootprint()
  File "C:\Users\uchan\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\.\fps.py", line 403, in addfootprint
    pcb = kicad_parser.KicadFcad(filename,via_skip_hole=False,via_bound=0)
  File "C:\Users\uchan\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\.\kicad_parser.py", line 632, in __init__
    self.part_path = getKicadPath(self.path_env)
  File "C:\Users\uchan\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\.\kicad_parser.py", line 497, in getKicadPath
    content = f.read()

'cp932' codec can't decode byte 0x88 in position 604: illegal multibyte sequenceQWindowsWindow::setMouseGrabEnabled: Not setting mouse grab for invisible window QWidgetWindow/'QWhatsThatClassWindow'
```

I could fix this error by changing this line

```
﻿    with open(kicad_common,'r') as f:
```

as follows

```
    with open(kicad_common,'r', encoding='utf-8') as f:
```

I suspect that newer versions of KiCad will use UTF-8, even when running in a Windows environment.

easyw, 2024-02-12:
@uchan-nos 
could you please attach a zipped footprint, which is failing the loading process?

easyw, 2024-02-13:
@uchan-nos 
I think is because of the path ${UCHAN_3DMODEL_DIR} in your kicad_mod fp.
The parser should be open the file coding in utf-8 anyway.
Thanks for finding the issue.
I'm going to push a fix as suggested.

easyw, 2024-02-13:
done https://github.com/easyw/kicadStepUpMod/commit/7028cac92b5d71d5d3ca2971b841bc9c64a0e4e7

uchan-nos, 2024-02-13:
Thank you for your quick response.

For your information:
UCHAN_3DMODEL_DIR = `C:\Users\uchan\SynologyDrive\electronics\kicad_library\3dmodels`.
There are no spaces or non-ASCII characters.

easyw, 2024-02-15:
@uchan-nos 
it is strange that I couldn't replicate the issue, anyway the parser now is opening the file coding in utf-8.
I'm closing this as solved.
Thanx.

closed 2024-02-15 by easyw