easyw/kicadStepUpMod #210: Support KiCAD 8.0

thoni56, 2024-03-06:
I'm not completely sure, but it seems that StepUpMod crashes on KiCad 8.0 files:

`16:38:42  Running the Python command 'ksuToolsOpenBoard' failed:
Traceback (most recent call last):
  File "C:\Users\Thomas\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\.\kicadStepUpCMD.py", line 1186, in Activated
    kicadStepUptools.onLoadBoard()
  File "C:\Users\Thomas\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\.\kicadStepUptools.py", line 6699, in onLoadBoard
    modules,nsk = DrawPCB(mypcb,SketchLayer,override_pcb,keep_pcb_sketch)
  File "C:\Users\Thomas\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\.\kicadStepUptools.py", line 12338, in DrawPCB
    line.append(m.fp_text[0][1]) #fp reference
  File "C:\Users\Thomas\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\.\fcad_parser\sexp_parser\sexp_parser.py", line 147, in __getitem__
    v = self._value[key]

list index out of range`

I didn't realize that 8.0 was not supported so I don't have the same model in KiCAD 7.0 to compare to.

[thatreus20d44k.kicad_pcb.txt](https://github.com/easyw/kicadStepUpMod/files/14512396/thatreus20d44k.kicad_pcb.txt)

easyw, 2024-03-07:
it seems you have an old version of kSU...
could you please post your FC full version info and kSU full version info?

thoni56, 2024-03-07:
I have upgraded and am now running FreeCAD 0.21.2 with kSU 10.23.2 but get the same error.

easyw, 2024-03-07:
could you please to delete the workbench and then re-install it?
Sometimes FC Tools is not updating the wb correctly.
I have tested your board with FC0.21.2 and FC0.22 dev without issues
w/ KiCadStepUpWB v 10.23.2 and kicad StepUp version 11.1.7

thoni56, 2024-03-07:
Yes, removing, restarting and installing a-fresh did the trick. Thanks.

closed 2024-03-07 by thoni56