easyw/kicadStepUpMod #213: Push 3d moved model to PCB fails for KiCad 8

0xxon, 2024-04-08:
Hi,

when trying to push a moved 3d model to a KiCad 8 PCB, the process fails with the message "Select only 3D model(s) moved to be updated/pushed to kicad board! a Time Stamp is required!"

The same process works when it is instead done using an equivalent PCB file created in KiCad 7. I assume that something slightly changed in the file format that breaks this (as it previously happened in #75). 

The other 3d model syncing commands fail with similar messages.

0xxon, 2024-04-08:
oh - and I just saw that this is mentioned in the readme and I didn't note it in time. In that case - you can just close this, or leave it open in case anyone else stumbles across the same issue.

onazir, 2024-05-18:
Hi,
[image]
Oxxon,  I didn't find anything on "readme" 

push 3D moved function still doesn't work.

easyw, 2024-06-04:
@onazir 
[image]

Unfortunately I have very little time ATM to fix it and I'm working on a previus relase of KiCAD.
Any PR to fix it would be welcomed.

easyw, 2024-06-19:
@onazir @0xxon it should be now fixed
https://github.com/easyw/kicadStepUpMod/commit/20458d0979ad66735b9674e79933f1f5a0215e81

closed 2024-06-19 by easyw