easyw/kicadStepUpMod #215: Invalid ambientIntensity value for GLASS-19 material

adamalfath, 2024-04-16:
The issue was found by another developer (https://github.com/fougue/mayo/issues/268) while I was trying a 3d viewer program.

From quick reading, the ambientIntensity value per VRML 2.0 specification should be in the range of 0.0 to 1.0?

easyw, 2024-04-16:
@adamalfath 
could you please post a sample of the vrml file?
which is the wrong value?

adamalfath, 2024-04-16:
Here is the example file: [LCM_ZJY096T-IG11.zip](https://github.com/easyw/kicadStepUpMod/files/14990700/LCM_ZJY096T-IG11.zip) line 270 `ambientIntensity 2.018212`

Since the material definition is automatically generated from KSU, it will be present on every VRML output file.

easyw, 2024-04-16:
thanks, but which is the correct ambientIntensity value to be applied?

closed 2024-04-28 by easyw