easyw/kicadStepUpMod #217: PySide2 deprecated, addon does not work on Archlinux

Kezii, 2024-05-02:
`02:10:54  During initialization the error "No module named 'PySide2'" occurred in /home/[...]/.local/share/FreeCAD/Mod/kicadStepUpMod/./InitGui.py`

Apparently PySide2 is EOL and works only with older versions of python

https://code.qt.io/cgit/pyside/pyside-setup.git/log/?h=5.15.2
https://wiki.qt.io/Qt_for_Python

Project should migrate to PySide6

easyw, 2024-05-02:
hi @Kezii 
which is your kSU version?
Anyway, following your issue I searched for all instances of PySide2 and I removed all of them with the latest commit...

Kezii, 2024-05-02:
the latest installing with the addon manager, now I updated and the addon appears, thank you

closed 2024-05-02 by Kezii