easyw/kicadStepUpMod #219: Exporting 3D Model fails when there are parts in parts

TimGoll, 2024-05-16:
[image]

I can't export the "picolibro" part here, because the sub parts don't have a shape color attribute. The elements within this part are colored though. I think this has something to do with the VWRL exporter.

```
16:15:17  Running the Python command 'ksuToolsExportModel' failed:
Traceback (most recent call last):
  File "C:\Users\tigo\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod.backup1701772747.179403\.\kicadStepUpCMD.py", line 1285, in Activated
    routineScaleVRML()
  File "C:\Users\tigo\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod.backup1701772747.179403\.\kicadStepUptools.py", line 7416, in routineScaleVRML
    lbl=go_export(fullFilePathName)
  File "C:\Users\tigo\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod.backup1701772747.179403\.\kicadStepUptools.py", line 2684, in go_export
    export(objs, fPathName, 0.3937, lbl)
  File "C:\Users\tigo\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod.backup1701772747.179403\.\kicadStepUptools.py", line 2298, in export
    color.append(FreeCADGui.ActiveDocument.getObject(obj.Name).ShapeColor)

'Gui.ViewProviderDocumentObject' object has no attribute 'ShapeColor'
```

## Versions

- KiCad Stepup: V11.1.5
- FreeCad v0.21.2

easyw, 2024-06-19:
@TimGoll 
in kSU you can use he make union or make compound and then export it

TimGoll, 2024-06-19:
> @TimGoll 
> in kSU you can use he make union or make compound and then export it

But that looses all existing colors in the model

easyw, 2025-05-22:
@TimGoll , sorry for delay in replying.
To export as step or wrl, you can use the FC standard exporter method.