easyw/kicadStepUpMod #220: Can't Get Combo View

mampir, 2024-05-21:
Hello,

When I try to `Export 3D Model` or another action, I get the following error log:

<pre>
PoM not present
11:38:43  using 'Part' container and 'Links'
11:38:43  FC Version 021-2
11:38:43  kicad StepUp version 12.0.3
11:38:43  tolerance on vertex applied
11:38:43  applying Materials to Shapes
11:38:43  your home path is /home/mmp
11:38:43  kicadStepUp
11:38:43  closed
11:38:43  Running the Python command 'ksuToolsExportModel' failed:
Traceback (most recent call last):
  File "/home/user/.local/share/FreeCAD/Mod/kicadStepUpMod/./kicadStepUpCMD.py", line 1289, in Activated
    from kicadStepUptools import routineScaleVRML
  File "/home/user/.local/share/FreeCAD/Mod/kicadStepUpMod/./kicadStepUptools.py", line 21781, in <module>
    KSUWidget.activateWindow()
    ^^^^^^^^^

name 'KSUWidget' is not defined
</pre>

I'm using FreeCAD 0.21.2 and KiCadStepUp revision 56df083ae33473b5eb7698307ed181cb5c771032.

mampir, 2024-05-21:
I managed to make the export button work with this ugly thing:

<pre>
diff --git a/kicadStepUptools.py b/kicadStepUptools.py
index 7ae55fe..2d86899 100644
--- a/kicadStepUptools.py
<pre>
+++ b/kicadStepUptools.py
@@ -21732,6 +21732,12 @@ if singleInstance():
         cv = t.findChild(QtGui.QDockWidget, "Model")
         if cv is None:
             cv = t.findChild(QtGui.QDockWidget, "Tree view")
+            if cv is None:
+                cv = [o for o in t.children () if o.objectName () == "Combo View"]
+                if cv:
+                    cv = cv[0]
+                else:
+                    cv = None
     #say( "Combo View" + str(cv))
     ## print( "KSUWidget" + str(wf))        
     cv.setFeatures( QtGui.QDockWidget.DockWidgetMovable | QtGui.QDockWidget.DockWidgetFloatable|QtGui.QDockWidget.DockWidgetClosable )
</pre>

Should I make a pull request this?

easyw, 2024-05-23:
> Should I make a pull request for this?

yes please

mampir, 2024-05-23:
The pull request is here: https://github.com/easyw/kicadStepUpMod/pull/221

easyw, 2024-05-31:
hi @mampir 
I annot reproduce your issue...
could you please attach a zipped version of your FreeCAD _'user.cfg'_ file?

mampir, 2024-05-31:
[user.cfg.zip](https://github.com/user-attachments/files/15518413/user.cfg.zip)
Here you go. You may also note, that I'm using I'm using this packaging of FreeCAD: https://www.parabola.nu/packages/libre/x86_64/freecad/

easyw, 2024-06-19:
@mampir 
I cannot reproduce your issue...
it could be related to your distribution/packaging...
I'm closing it for the moment

closed 2024-06-19 by easyw

mampir, 2024-06-19:
That's very disappointing and annoying. Of course it is because of different distribution and packaging - people use more than one distribution and packaging in this world. You could have just implemented the simple fix I already submitted.

easyw, 2024-08-26:
@mampir 
it should be now implemented with https://github.com/easyw/kicadStepUpMod/commit/b5ef27d2f4eaaa4bf60be5339ba7787408462979

mampir, 2024-09-12:
Thank you for implementing this! Unfortunately I'm not sure if this is necessary anymore, because the current version of FreeCAD that I have doesn't have these issues anymore. It seems to the package maintainers applied some patches to FreeCAD which deal with these issues. I think that was done on July 27, but I'm not sure.