easyw/kicadStepUpMod #225: Add ability to import and render the SolderMask layers

JimBach-IMMI, 2024-07-18:
kicadStepUp is a really nice tool . . . love it . . . except that it can't import/render the SolderMask.
It's important to be able to verify that solder mask is open where you need it open (such as Test Points), and it's nice to have the solder mask present when you are trying to do a photo-realistic 'render' for reports and presentations.  

Please add the ability to import/render the Solder Mask, even if it means exporting it from KiCAD in DXF format (like the Silk Screens).

Thanks (in advance)

easyw, 2024-07-19:
@JimBach-IMMI 
you can already export the SolderMask in DXF and load it using the same button for the Silks Screen.
Then just rename the shape to SolderMask and change its color.
Moreover kv8.99 seems is supporting also Soler Mask export

JimBach-IMMI, 2024-07-19:
> [CAUTION: This email originated from outside of IMMI. Do not click links or open attachments unless you recognize the sender's email address and know the content is safe.]

Thanks for the quick response to this.  OK, I was pondering trying that, but I wasn’t sure if it was going to overwrite the existing SilkScreen.  Glad to know that it won’t . . . I’ll give it a try.
How does the software know what “Z” level (height above the bare board) to put the Silks and Solder Masks? Generally Solder Mask needs to be above Copper, and Silks needs to be above Solder Mask.

Perhaps the “AddSilks” button could be renamed “AddSilksSolderMask” or something . . . or have 2 separate buttons.  That would make it much more obvious to newbie users (like me) that you CAN get the Solder Mask that way😊

So, why is it that the SilkScreen and SolderMask have to be imported via DXFs and can’t be sucked-in from the _PCB file (like board and copper)??

Also . . . What is “kv8.99”??

Thanks Again,

Jim Bach  |  IMMI  | Principal Electrical Engineer
18881 IMMI Way | Westfield, IN 46074
Office: 317-867-8119 | Mobile: 317-507-6724
imminet.com<http://www.imminet.com/>

This correspondence is intended solely for the person or entity to which it is addressed and contains confidential 
information.

This email has been scanned for viruses and malware by Mimecast Ltd, an innovator in Software as a Service (SaaS) for business.

JimBach-IMMI, 2024-07-19:
> OK . . . I exported ALL of the KiCAD layers to DXF and then tried using AddSilks to bring-in the Solder Mask layers:
>
> And StepUp created:
>
> But when I select the new top solder mask I see:
>
> The mask file (also see attached) looks like:
>
> It’s almost like StepUp put the mask where the openings in the mask are supposed to go . . . the inverse of what I want.  Is there some option I’m missing (either in KiCAD when exporting to DXF or in StepUp)??
>
> Regards,

I did some digging . . . in KiCAD’s export menu there’s an option:
***@***.***

So, I regenerated the DXFs again . . .
Now when I import the Solder Masks I see:
***@***.***

And reimporting the Silk Screens I now see:
***@***.***

☹

Regards,

Jim Bach  |  IMMI  | Principal Electrical Engineer
18881 IMMI Way | Westfield, IN 46074
Office: 317-867-8119 | Mobile: 317-507-6724
imminet.com<http://www.imminet.com/>

easyw, 2024-07-19:
@JimBach-IMMI 
the SolderMask is exported in 'negative' way.
You may obtain the real Solder Mask with a 'Cut' from the pcb face.
Anyway kv8.99 (kicad version 8.99) is able to export tracks, pads, silks and solder 
https://forum.kicad.info/t/post-v8-new-features-and-development-news/48614/29

JimBach-IMMI, 2024-07-19:
Guess I’m not sure what you mean a “’Cut’ from the PCB face” . . . I don’t normally use FreeCAD . . . just using it to get from KiCAD to SolidWorks (STEP).
I’m tempted to install 8.99 . . . but scared at the same time . . .

Regards,

Jim Bach  |  IMMI  | Principal Electrical Engineer
18881 IMMI Way | Westfield, IN 46074
Office: 317-867-8119 | Mobile: 317-507-6724
imminet.com<http://www.imminet.com/>

Anyway kv8.99 (kicad version 8.99) is able to export tracks, pads, silks and solder mask
https://forum.kicad.info/t/post-v8-new-features-and-development-news/48614/29<https://forum.kicad.info/t/post-v8-new-features-and-development-news/48614/29>

closed 2025-01-13 by easyw