easyw/kicadStepUpMod #226: Some PTH Vias Are Not Present in Rendering

JimBach-IMMI, 2024-07-18:
Here's a weird one for you; I hope you can solve it easily. 

I have a PCB that has a plethora of via holes that are part of "footprints" (via-in-pad of an LED module, vias around a mounting hole to add no-crush feature) that DO show-up in the StepUp import (and rendering) of a KiCAD design, and I also have a plethora of vias that were added to the design using the "Via" icon in the KiCAD PCB toolbar that do NOT show-up in the StepUp import (rendering).  Thus, when I export the StepUp model of my board to STEP and import it into my CAD tool those holes are not present, even though the copper pattern shows that there should be holes there.

Those vias ARE present in the PTH DrillMap Gerber file, so KiCAD knows they are there and exports them in the Gerber-making process.  But when StepUp imports the "Board" (directly from the _PCB file) it somehow doesn't know about those drilled holes.

easyw, 2024-07-19:
Hi,
via drills are omitted by design, to avoid increasing in loading time and file size result.
Anyway you can enable vias in kSU settings
[image]

closed 2024-07-19 by easyw

JimBach-IMMI, 2024-07-19:
Many thanks for the quick and informative response to this.  Cool . . . glad to see that it was already there.  Enabled it as shown, and tested it . . . works like a charm!!  Thanks!