easyw/kicadStepUpMod #228: Licensing and bundling

prokoudine, 2024-09-26:
Hi Maurice, sorry about using the public tracker, but my email probably went into your spam folder.

We (Ondsel) are interested in shipping the workbench as part of Ondsel ES. It looks like our variation of FreeCAD being licensed under LGPL is not a problem as it can incorporate AGPL code (IANAL). But then there's the topic of trademarking, and this is where I would really appreciate your guidance.

easyw, 2024-10-01:
Hi @prokoudine,
sorry but I rarely check my account.
We can discuss here or at the FC forum if you prefer.
1) GPL vs AGPL: 
GPL  has one added requirement: if you run a modified program on a server and let other users communicate with it there, your server must also allow them to download the source code corresponding to the modified version running there.

I don't know Ondsel well enough...
May I ask you some info?
a) I see you give a free license for working offline (Solo option)... is it a 'forever' free license option or your plan could change?
b) A user of the Solo plan can install extra WorkBenches?
c) A user of the Solo plan can run custom Python scripts?

prokoudine, 2024-10-01:
Here is fine for me. 

The tiers currently cover the cloud service (Ondsel Lens), not the desktop software (Ondsel ES). You do need a Lens account to download the desktop software, but the software itself is free to download and use. It's also under the same LGPL2.1+ license as the vanilla FreeCAD, with the same Addon Manager, so there are no llimitations over installing and running 3rd-party addons. All our changes so far are available publicly, most have been upstreamed and merged to FreeCAD.

It's possible that, going forward, specifics of what we offer in the paid tier could change. We could adjust the pricing, we could offer builds of the desktop software with new extra features that are not available in the regular free build. But there would always be a free tier with a modification of FreeCAD available for downloading, under the same original license.

easyw, 2024-11-03:
Hi, I'm closing this issue because I just got aware that Ondsel is closing down.

closed 2024-11-03 by easyw