easyw/kicadStepUpMod #233: Freecad 1.0 RC2 incompatibility

9R, 2024-11-03:
KicadStepUp does not work with Freecad 1.0 any more. It seems some modules have changes:
```
14:36:50  KiCadStepUpWB.Activated(v 11.03.9)
14:36:53  PoM not present
14:36:53  using 'Part' container
14:36:53  Running the Python command 'ksuToolsOpenBoard' failed:
Traceback (most recent call last):
  File "/home/user/.local/share/FreeCAD/Mod/kicadStepUpMod/./kicadStepUpCMD.py", line 1205, in Activated
    import kicadStepUptools
  File "/home/user/.local/share/FreeCAD/Mod/kicadStepUpMod/./kicadStepUptools.py", line 808, in <module>
    import kicad_parser
  File "/home/user/.local/share/FreeCAD/Mod/kicadStepUpMod/./kicad_parser.py", line 23, in <module>
    import Path

No module named 'Path'
```

The `Path` module seems to have been renamed to `CAM`.

Changing the import in  `kicad_parser.py` fixes at least the `Import Board` and `Push Sketch to PCB` functions work after this change.

PR https://github.com/easyw/kicadStepUpMod/pull/232 restores basic functionality with FreeCAD 1.0

easyw, 2024-11-03:
Hi,
I cannot replicate your issue...
Could you please attach your full FC info?
Which is your starting WB?
Could you check if typing in the python console
`import Path`
you get errors?
I tried the same in FC and I don't trigger any error....

9R, 2024-11-04:
Thank you for your quick response and of course foremost for creating this workbench in the first place :)

# FC Versions
I have tested against various packages of FreeCAD 1.0RC2:
archlinux package [ FreeCAD 1.0.0, Libs: 1.0.0RC2R38806 ]
nixOS package [ FreeCAD 1.0.0, Libs: 1.0.0RC2RUnknown ]
official x86_64 app-image from [here](https://github.com/FreeCAD/FreeCAD/releases/tag/1.0rc2)

# Default WB
My starting WB is PartDesign. I have also tested loading other workbenches before kicadStepUp but the result is the same.

# Console
Running `import Path` in the console on any of the above versions results in a variation of `ModuleNotFoundError`:

## Arch:
```
>>> import Path
Traceback (most recent call last):
  File "<input>", line 1, in <module>
ModuleNotFoundError: No module named 'Path'
```

## nixOS
```
>>> import Path
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/nix/store/zh56w35s3g8b7ws2xxxlzl610cb9mgij-shiboken2-5.15.14/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ModuleNotFoundError: No module named 'Path'
```

## appimage directly from freecad.org
```
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/tmp/.mount_FreeCAbYVQal/usr/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ModuleNotFoundError: No module named 'Path'
```
# FC 0.21
Opposed to that, when using the arch-aur or the official [app-image](https://github.com/FreeCAD/FreeCAD/releases/download/0.21.2/FreeCAD-0.21.2-Linux-x86_64.AppImage) from freecad.org `import Path` works as expected.

When I compared 1.0 vs 0.21 file structure `Path` is indeed missing from the `Mod` dir.

9R, 2024-11-04:
related PR #232

9R, 2024-11-04:
PR #232 that resolves this issue has been merged. Thanks for the quick response. :)
Closing.

closed 2024-11-04 by 9R