easyw/kicadStepUpMod #236: "CAM" module import errors in RealThunder's Link branch

BraidenPsiuk, 2024-11-28:
Hi, I use [RealThunder's Link branch of FreeCAD](https://github.com/realthunder/FreeCAD) quite a bit, and wanted to try out kiCadStepUp for the first time today. I installed it via the Addon Manager and noticed I was getting some errors related to not finding the "CAM" module.

I saw that the recent 1.0 main branch release of FreeCAD kinda renamed `Path` to `CAM`, and noticed PR #232 added a conditional to handle this name change. I tested and saw it works in FreeCAD main branch just fine, but I was wondering if a small tweak could be done to get things loading properly in the Link branch.

If you run
```python
FreeCAD.Version()
```
in the Python console of the Link branch, it returns something like
```json
['2024', '1006', '0', '14555 (Git shallow)', 'Unknown', '20241006', '(HEAD detached at 20241003stable)', 'a9810d509a6f112b5ac03d4d4831b67e6bffd5b7']
```
so the else condition of the section that checks the version in `kicad_parser.py` is met and it tries to import `CAM`. However, the Link branch has not renamed `Path` to `CAM` yet.

Not sure if you provide support for branches of FreeCAD, and I understand it isn't really a bug with kiCadStepUp, but the Link branch is pretty popular and I would think it wouldn't require much changing to fix. If this is okay and you don't have the time I am happy to try my own PR- I noticed some info in `FreeCAD.ConfigDump()` could be helpful to check if it is the Link branch or not.

---

For the time being until this is handled, to anyone with the link branch installed as an Appimage who just wants a quick n' dirty patch to get the KiCadStepUp workbench working, just edit ~/.local/share/FreeCAD/Mod/kicadStepUpMod/kicad_parser.py and replace
```python
if FreeCAD.Version()[0] < '1':
```
with
```python
if FreeCAD.Version()[0] >= '1':
```
to force it to use the old module name.

easyw, 2024-11-28:
@BraidenPsiuk 
a solution could it be:
```
try:
    import Path
    CAM = Path
except:
    import CAM
```

BraidenPsiuk, 2024-11-29:
Yeah! Just tried that, it loaded successfully in the Link branch. Seems simple enough.

BraidenPsiuk, 2024-11-29:
Should probably be flipped though, right?
```python
try:
    import CAM
except:
    import Path
    CAM = Path
```

BraidenPsiuk, 2024-11-29:
#237

closed 2024-12-04 by easyw