easyw/kicadStepUpMod #238: Error pushing the sketch

dzid26, 2024-12-29:
```
19:49:45  evaluate to recompute
19:49:45  check to sanitize
19:49:48  Running the Python command 'ksuToolsPushPCB' failed:
Traceback (most recent call last):
  File "C:\Users\dzidm\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\.\kicadStepUpCMD.py", line 1500, in Activated
    FreeCADGui.ActiveDocument.getObject(sk_obj.Name).Autoconstraints = True
                                        ^^^^^^^^^^^

'NoneType' object has no attribute 'Name' 
```
or 
```
19:52:50  Running the Python command 'ksuToolsPushPCB' failed:
Traceback (most recent call last):
  File "C:\Users\dzidm\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\.\kicadStepUpCMD.py", line 1500, in Activated
    FreeCADGui.ActiveDocument.getObject(sk_obj.Name).Autoconstraints = True
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

'Gui.ViewProviderGeometryObject' object has no attribute 'Autoconstraints'
```

This happens if a sketch has a dependency. 
For example if it is attached to a plane. 
[image]
When I clear Attachment Support field, I can push the sketch again.

If the sketch uses External Geometry there issue also appears - which is a bigger shame. 

Freecad 1.0

easyw, 2024-12-30:
FC 1.0 has changed a lot with sketch wb.
I still have issues in my workflow that I haven't solved.
For this issue, you could copy the sketch and remove the dependencies to the copy, so you shouldn't have any issue in the original sketch.
Could you attach a FC file to test?
I will try to solve this for next release of kSU.

easyw, 2024-12-30:
I just tested with the latest kSU release and I don't have any issue if I select only sketch when the dialog appear.
I would definitely need a FC sample file.

dzid26, 2024-12-30:
Let's start with something simple.  
For example, 
1. I just created a new simple sketch on a plane.  [sketch on a plane.zip](https://github.com/user-attachments/files/18277015/sketch.on.a.plane.zip)
2. I press "Push sSketch to PCB" button.
3. "Object selection" window opens up, I press OK - (this is already a bad sign)
4. The sketch is copied in the model tree and nothing happens.

```
18:25:24  check to sanitize
18:25:25  Running the Python command 'ksuToolsPushPCB' failed:
Traceback (most recent call last):
  File "C:\Users\dzidm\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\.\kicadStepUpCMD.py", line 1500, in Activated
    FreeCADGui.ActiveDocument.getObject(sk_obj.Name).Autoconstraints = True
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

'Gui.ViewProviderGeometryObject' object has no attribute 'Autoconstraints'
```

If I clear reference plane, it works - aka, KicadStepUp asks for the line width.

easyw, 2024-12-30:
if you simply deselect "All" and then add "Sketch" and deselect the attach "Plane", the tool will export the sketch correctly.
I'm attaching a screencast video:

https://github.com/user-attachments/assets/9dfa7376-dde3-4c3f-895d-a51556b9e9b6

closed 2025-01-09 by easyw

dzid26, 2025-01-10:
Your workaround is good, but maybe the issue should stay opened?