easyw/kicadStepUpMod #239: Kicad StepUp tools menu broken

benjinne, 2024-12-31:
Hello, my menu interface isn't working as intended on my machine. It looks like this. It seems like a scaling issue.
[image]
I'm currently running freecad 1.0.0 and kicad 8 on ubuntu 22.04.
I'm curious if anyone else has ran into this issue and if it's something I can fix.

benjinne, 2024-12-31:
Edit:

I reset all settings in Freecad 0.21 and went back to Freecad 1.0.0 and now it's working. I'm not sure what changed.

benjinne, 2024-12-31:
I found the problem. It looks like it has to do with the theme. Freecad classic works, but switching to light or dark theme breaks the menu.

dzid26, 2025-01-06:
As a workaround, opendark / openlight from openthemes "Addon" work ok.

easyw, 2025-01-09:
hi,
thanks for reporting it.
I will have a look at the issue-

easyw, 2025-04-22:
fixed w/ https://github.com/easyw/kicadStepUpMod/commit/a3fdcec0d9dab13bba071e23b854391b2022b510

closed 2025-04-22 by easyw