easyw/kicadStepUpMod #244: ksuToolsPushMoved failes based on the value field

jbayer05, 2025-01-16:
Hello,

I have no deep understanding of the code therefore I can just report what I found - but unfortunately I have not a solution.
I stumbled over an issue which leads to an Python-Error in push3D2pcb:
At line 21822 a check is performed "if ('at' in ln)" which wrongfully resolves true in the case a component has 'at' in its value.

In the attached example board, I have an LED with value 'Status' which triggers the if and then the "ln.index(') ')" fails with "substring not found".
After renaming the LED-Value to something else (which does not contain 'at'), the position-push works as expected.
As this check appears at other places in the code, I assume there could be other issues.

In order to reproduce the issue:
- load the attached [Status-LED.kicad_pcb.zip](https://github.com/user-attachments/files/18443320/Status-LED.kicad_pcb.zip) (KiCAD v8.0.7) with KiCadStepUp
- select the LED
- use the "ksu Push 3D moved model(s) to PCB" tool
- select the "Status-LED.kicad_pcb"

thanks a lot for the extremely useful workbench!

easyw, 2025-01-16:
hi @jbayer05 
thanks for your detailed feedback...
I've pushed some code to make the parser a bit more robust...
Please have a try.

jbayer05, 2025-01-17:
Hi @easyw,

thanks for the quick fix - works for me.
Of course you can still fool it by changing the value accordingly (e.g. "St(atus") - but of course this is much more unlikely now ;-)

Greetings

closed 2025-04-12 by easyw