easyw/kicadStepUpMod #247: STEP file generated by newest version not showing in KiCAD

der-ule, 2025-03-13:
I download some models from JST and used KSU to orient them and align them correctly, after exporting, I got my .step and .wrl files, however, after selecting the new files for my footprint, they are not visible in KiCad, the .wrl files show fine, I recently updated to the latest commit version as prompted by FreeCad, before that I exported some files without problem (ok, they were uniform gray, but no biggie). I am attaching one of the original files and its reworked version. 

[B04B-PASK.zip](https://github.com/user-attachments/files/19226848/B04B-PASK.zip)

[ksu_jst_b04b.zip](https://github.com/user-attachments/files/19226855/ksu_jst_b04b.zip)

KSU Version:
```
11:06:32  KiCadStepUpWB.Activated(v 11.05.5)
11:06:32  https://github.com/easyw/kicadStepUpMod-> commits:698
11:06:32  the WB is Up to Date
```

FreeCAD Version:
```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Assembly4 0.50.17
  * Assembly4.backup1741681054.9177063 0.50.15 (Disabled)
  * fasteners 0.5.37
  * kicadStepUpMod 11.5.5
  * Manipulator 1.5.7
```

KiCad Version:
```
Application: KiCad PCB Editor x64 on x64

Version: 8.0.9-3-gc4009544a9, release build

Libraries:
	wxWidgets 3.2.6
	FreeType 2.13.3
	HarfBuzz 10.0.1
	FontConfig 2.15.0
	libcurl/8.10.1-DEV Schannel zlib/1.3.1

Platform: Windows 10 (build 19045), 64-bit edition, 64 bit, Little endian, wxMSW
OpenGL: ATI Technologies Inc., AMD Radeon (TM) Graphics, 4.6.0 Compatibility Profile Context 24.9.1.240813

Build Info:
	Date: Feb 19 2025 04:21:22
	wxWidgets: 3.2.6 (wchar_t,wx containers)
	Boost: 1.86.0
	OCC: 7.8.1
	Curl: 8.10.1-DEV
	ngspice: 44
	Compiler: Visual C++ 1942 without C++ ABI

Build settings:
```

easyw, 2025-03-15:
@der-ule 
thanks for this report ...
This seems an issue in FreeCAD... moreover it is a sneaky issue because re-importing the exported file in FC, the issue is hidden.
kSU when is exporting a 3D model to wrl and step, is assigning color to model faces.
The process seems correct, in fact showing the appearance per face of the model, the transparency is set to the right value (0%).
Unfortunately when FC is exporting the file to step format, is assigning transparency to faces to 100% even if the value is 0%.
The sneaky part is that when FC is importing the file, will assign transparency to 0% instead to 100%.
Opening instead the exported file in a different CAD i.e. CAD Assistant, the model will be completely invisible, because of the 100% transparency assigned by FC to the faces.
(if you open the step file with a text editor, and search for "SURFACE_STYLE_TRANSPARENT(" the assigned value is "1."
This seems related to the big refactoring of FC, implementing the Material Appearance, introducing bugs in step importing/exporting routines.

easyw, 2025-03-20:
@der-ule 
could you please chek if this commit https://github.com/easyw/kicadStepUpMod/commit/414edc57a79115142a26bfc54040d1730d2e7ec1 is fixing it?

der-ule, 2025-03-28:
Updating to the newest version available the problem seems to be fixed

```
11:18:02  saveSettings Helper
11:18:02  PoM not present
11:18:02  using 'Part' container and 'Links'
11:18:02  FC Version 10-0
11:18:02  kicad StepUp version 12.6.0
```

[ksu_jst_b04b_v12.6.0.zip](https://github.com/user-attachments/files/19503648/ksu_jst_b04b_v12.6.0.zip)

looking at the diff between both files, it is as you explained before, the "SURFACE_STYLE_TRANSPARENT" value is not set to 1 anymore but to 0.

```
#2541 = SURFACE_STYLE_TRANSPARENT(0.);
#2550 = SURFACE_STYLE_TRANSPARENT(0.);
#2559 = SURFACE_STYLE_TRANSPARENT(0.);
#2568 = SURFACE_STYLE_TRANSPARENT(0.);
#2577 = SURFACE_STYLE_TRANSPARENT(0.);
#2586 = SURFACE_STYLE_TRANSPARENT(0.);
#2595 = SURFACE_STYLE_TRANSPARENT(0.);
#2604 = SURFACE_STYLE_TRANSPARENT(0.);
#2613 = SURFACE_STYLE_TRANSPARENT(0.);
#2622 = SURFACE_STYLE_TRANSPARENT(0.);
#2631 = SURFACE_STYLE_TRANSPARENT(0.);
#2640 = SURFACE_STYLE_TRANSPARENT(0.);
#2649 = SURFACE_STYLE_TRANSPARENT(0.);
#2658 = SURFACE_STYLE_TRANSPARENT(0.);
#2667 = SURFACE_STYLE_TRANSPARENT(0.);
#2676 = SURFACE_STYLE_TRANSPARENT(0.);
```

Thanks a lot for your great dedication, big fan of your software !

closed 2025-03-28 by der-ule