easyw/kicadStepUpMod #248: exception: key "layer" not found  for kicad v9 board file

nickoe, 2025-04-01:
Hello @easyw  

Thank you very much for your great effort and dedication on your lovely KSU tool!

I created a board in KiCad v9, specifically Version: 9.0.1-rc1-13-g029f72161b, release build Date: Mar 18 2025 05:30:14 . I also have v8 installed as Version: 8.0.7, release  Date: Dec 3 2024 15:58:01 build . Testing on windows. 

I am not sure which version of kicad if any it detects if it uses the python scripting interface at all. 

I get this exception and some "invalid Boolean data" messages that does not appear like something that needs to be updated in some parser.

```
14:39:48  PoM not present
14:39:48  using 'Part' container and 'Links'
14:39:48  FC Version 10-0
14:39:48  kicad StepUp version 12.6.0
14:39:48  tolerance on vertex applied
14:39:48  applying Materials to Shapes
14:39:48  your home path is C:\Users\myuser
14:39:48  C:\Program Files\KiCad\10.0\share\kicad\3dmodels
14:39:48  C:\Program Files\KiCad\9.0\share\kicad\3dmodels
14:39:48  export to STEP False
14:39:48  Tabified done !
14:39:48  C:\Program Files\KiCad\10.0\share\kicad\3dmodels
14:39:48  C:\Program Files\KiCad\9.0\share\kicad\3dmodels
14:39:48  done!
14:39:48  default light direction applied
14:40:09  opening C:/Users/myuser/Documents/MyProject/MyProject.kicad_pcb
14:40:09  my file path C:\Users\myuser\Documents\MyProject
14:40:09  opening Transaction 'opening_kicad'
14:40:09  ('invalid boolean data', [4160, 'locked', 'yes'])
14:40:09  ('invalid boolean data', [4366, 'locked', 'yes'])
14:40:09  ('invalid boolean data', [10794, 'locked', 'yes'])
14:40:09  ('invalid boolean data', [16255, 'locked', 'yes'])
14:40:09  ('invalid boolean data', [19013, 'locked', 'yes'])
14:40:09  ('invalid boolean data', [24329, 'locked', 'yes'])
14:40:09  ('invalid boolean data', [32741, 'locked', 'yes'])
14:40:09  ('invalid boolean data', [33865, 'locked', 'yes'])
14:40:09  ('invalid boolean data', [34585, 'locked', 'yes'])
14:40:09  ('invalid boolean data', [38198, 'locked', 'yes'])
14:40:09  ('invalid boolean data', [47002, 'locked', 'yes'])
14:40:09  ('invalid boolean data', [49220, 'locked', 'yes'])
14:40:09  ('invalid boolean data', [51329, 'locked', 'yes'])
14:40:09  ('invalid boolean data', [51661, 'locked', 'yes'])
14:40:09  ('invalid boolean data', [56910, 'locked', 'yes'])
14:40:09  ('invalid boolean data', [59637, 'locked', 'yes'])
14:40:09  ('invalid boolean data', [62216, 'locked', 'yes'])
14:40:09  ('invalid boolean data', [64775, 'locked', 'yes'])
14:40:09  ('invalid boolean data', [64892, 'locked', 'yes'])
14:40:09  ('invalid boolean data', [72314, 'locked', 'yes'])
14:40:09  grid origin not set
using default top left corner
14:40:09  PCB Loader 
14:40:09  pcb thickness 1.6mm
14:40:09  kicad_pcb version 20241229
14:40:09  parsing
14:40:09  Running the Python command 'ksuToolsOpenBoard' failed:
Traceback (most recent call last):
  File "C:\Users\myuser\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\.\kicadStepUpCMD.py", line 1212, in Activated
    kicadStepUptools.onLoadBoard()
  File "C:\Users\myuser\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\.\kicadStepUptools.py", line 6955, in onLoadBoard
    modules,nsk = DrawPCB(mypcb,SketchLayer,override_pcb,keep_pcb_sketch)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\myuser\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\.\kicadStepUptools.py", line 11978, in DrawPCB
    if lyr in ln.layer:
              ^^^^^^^^
  File "C:\Users\myuser\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\.\fcad_parser\sexp_parser\sexp_parser.py", line 180, in __getattr__
    raise AttributeError('{}: key "{}" not found'.format(self._line,name))

75759: key "layer" not found
```

I am not able to provide a test file at the moment, but I assume it is easy to reproduce. I could try to provide a sample later if needed.

Freecad version:
```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/Denmark (en_DK)
Stylesheet/Theme/QtStyle: Dark.qss/unset/Qt default
Installed mods: 
  * A2plus 0.4.68
  * CurvesWB 0.6.39
  * ExplodedAssembly
  * FreeCAD_FastenersWB 0.5.21
  * kicadStepUpMod 11.5.7
```

nickoe, 2025-04-01:
It looks like there is a log of board file version updates in:
https://docs.kicad.org/doxygen/pcb__io__kicad__sexpr_8h_source.html

And I am able to load an older board with version 20241228

easyw, 2025-04-02:
hi @nickoe 
I think I have found the culprit, but having a failing pcb file would be faster and easier.

nickoe, 2025-04-02:
Thank you, I was just trying to learn to use pdb... to debug the addon, but it is really hard for me to do manually, and I am not reeaallly sure what these sexpr things are:
```
22:15:40  (Pdb) odict_keys(['start', 'end', 'stroke', 'layers', 'net', 'uuid'])
22:15:46  (Pdb) odict_keys(['start', 'end', 'stroke', 'layer', 'uuid'])
22:15:52  (Pdb) odict_keys(['start', 'end', 'stroke', 'layer', 'uuid'])
```
from printing some random indexes of `print(mypcb.gr_line[1])`... I can't figure out how to do multi line expressions in the console with pdb imported.

easyw, 2025-04-03:
Hi,
I think your board could have some item of the new format option:
_`"Graphic shapes can have soldermask layer and margin"`_
You can check if you have lines similar to:
`	(gr_line
		(start 101 94)
		(end 164.5 63)
		(stroke
			(width 0.05)
			(type solid)
		)
		(layers "F.Cu" "F.Mask")
		(solder_mask_margin 1)
		(uuid "98b1afad-8124-4936-974f-ab95c89b9ec3")`
or in alternative with these items:
`                'gr_line',
                'gr_circle',
                'gr_arc',
                'gr_poly',
                'gr_curve',
                'gr_rect'`

nickoe, 2025-04-03:
There is gr_rect with F.Cu and F.Mask

easyw, 2025-04-03:
@nickoe 
it should be fixed w/ https://github.com/easyw/kicadStepUpMod/commit/fa96c0efd389c43a77cae7f544ca942651989d57

nickoe, 2025-04-03:
@easyw  yes! I can confirm it works with that commit.   I still see those `'invalid boolean data'` messages. I am not sure if they are important or not? Or maybe it should be tracked in another issue if that is important?

closed 2025-04-03 by nickoe

easyw, 2025-04-04:
> Or maybe it should be tracked in another issue if that is important?

yes, please open a new issue.
Fortunately, this is not a blockage, but it should be handled smoothly.