easyw/kicadStepUpMod #249: pushing edges from FC to KC - corners radiuses are mirrored

glbarak, 2025-04-07:
the pictures explain the problem:

a sketch in freecad:
[image]

resulting edge in Kicad

[image]

easyw, 2025-04-07:
@glbarak 
please post:
- FC version
- Kc version
- kSU version
- a file with the issue

these can help in solve the issue

glbarak, 2025-04-07:
FC 1.0.0
KC 8.0.8
KSU 11.6.0

Cannot attach FCStd file - it says that this kind of file is not allowed

glbarak, 2025-04-07:
Just to make sure I also checked with KC 9.0.1 - same thing

easyw, 2025-04-07:
@glbarak 
simply zip the file, it would be then allowed

glbarak, 2025-04-07:
Attached

[File_With_Problem.zip](https://github.com/user-attachments/files/19633088/File_With_Problem.zip)

easyw, 2025-04-07:
it is an issue related to the fact your sketch is inside a Body...
it should be 'workarounded' now with 
https://github.com/easyw/kicadStepUpMod/commit/dd772c4a08a4e40ef392516164fd7e4baafab8a0

glbarak, 2025-04-08:
I did not understand what to do with the workaround files.
But if I understand correctly taking the sketch out of the body should solve this. My "normal" workflow is part design WB with a body but I have no problem using sketcher only

easyw, 2025-04-08:
This version should work also w/ a sketch inside a body

glbarak, 2025-04-08:
I updated the WB
Before you posted I already removed the body so now there is only a sketch in the file
However the issue persists
Attached  the new file with no body

[FIle_With_Problem.zip](https://github.com/user-attachments/files/19648387/FIle_With_Problem.zip)

glbarak, 2025-04-08:
by the way - the updated version identifies differently in he title and in the"installed version"

[image]

easyw, 2025-04-08:
You probaby nerd to refresh the fc tool cache

glbarak, 2025-04-08:
Refreshed cache in the add-on manager screen - Still same behavior

easyw, 2025-04-08:
Then it should be a FC tool issue, not kSU WB.

easyw, 2025-04-12:
The edges can be inverted when manipulating Fc sketches.
Anyway with your user case it should be solved.

I'm closing the issue. Fell free to reopen it in case you may need it.

closed 2025-04-12 by easyw