easyw/kicadStepUpMod #250: KiCad 8 | FreeCad v1.0 |  Built in KiCad 3D models (xyz.step files) looks transpanent on the exported 3D step file

csbflx, 2025-04-11:
[image]

[image]

Hello DevTeam!

Can you please check why the original KiCad 3D objects looks transpanent after exporting the pcba to step? Other objects which downloaded for example from grabcad.com ( colored leds, inductor, relay, orange connectors) looks Ok.

Thank you in advance
CsabaD

easyw, 2025-04-12:
@csbflx 
could you please add:
FC full version
Kc full version
StepUp version
Do you have some step file and Fc file to post here?
(you would need to zip the files to upload to Gh)

csbflx, 2025-04-14:
[image]
[image]

[KiCad_StepUp_ExportedStep_TranspanentObjects_Issue.zip](https://github.com/user-attachments/files/19731058/

KiCad_StepUp_ExportedStep_TranspanentObjects_Issue.zip)

Hi DevTeam!

As you requested I send you the required info:

FREECAD v1.0

OS: Windows 10 build 17763
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda

KICAD v8.0

Application: KiCad x64 on x64

Version: 8.0.3, release build

Installed mods: 
  * kicadStepUpMod 11.6.2

I prepared a small size project to help you to reproduce this issue. It's included the exported step file with the transpanent KiCad resistor object (which comming from the KICAD8_3DMODEL_DIR), and the external downloaded content which looks ok. (Connector and Led). I attached also in this project folder a NIST STEP File Analyzer report html. (StepUp_Export-sfa.html)

"Did you exported the file with Fc std export menu? Or with kSU exporter tool?"
I exported the file with FC std export menu.

Hope its help for you, In case of any request feel free ask me.

Thank you very much again for the support and this great tool
CsabaD

easyw, 2025-04-14:
@csbflx 
could you please check the file? I cannot download it

csbflx, 2025-04-14:
[KiCad_StepUp_ExportedStep_TranspanentObjects_Issue.zip](https://github.com/user-attachments/files/19732534/KiCad_StepUp_ExportedStep_TranspanentObjects_Issue.zip)

csbflx, 2025-04-14:
I dont know it help for you:

In the meantime I had an idea: I have an old FreeCad installer v0.21.0. So I installed this old revision and tried the same. 
Everything became OK. I checked also the KCStepup version, its the same latest: v11.6.2.

I'm not sure but until now it seems there is a FC bug in the v1.0

[image]

easyw, 2025-04-14:
could you also please add a FreeCAD 1.0 file from which you have exported the step?

csbflx, 2025-04-14:
I prepared the file for both FC revisions ( FC v1.0 And FC v0.21.0

[KiCad_StepUp_ExportedStep_TranspanentObjects_Issue.zip](https://github.com/user-attachments/files/19732896/KiCad_StepUp_ExportedStep_TranspanentObjects_Issue.zip)

csbflx, 2025-04-14:
Hello

Until the fix arrives , I would propose for  the users "as a workaround" to download/install the FreeCad v0.21.0 version, then add the latest release of KiCadStepUpMod ( today is the 11.6.2 is the latest). Do the same as before.
For justification I attached the solution which generated by the above mentioned way.

[image]

Hope the solution/fix will be easy, 

Thank you in advance, greetings for all KicadStepUp dev team member. I really appreciated this tool!
CsabaD

easyw, 2025-04-14:
@csbflx 
could you please try if this commit is solving the issue?
https://github.com/easyw/kicadStepUpMod/commit/37c1acdb055ac9adb554623f38e9a44f452a9343

Note:
1. the fix works for FC 1.0.0 ONLY
2. to apply and fix the exporting issue you need to use the kSU export button

[image]

csbflx, 2025-04-15:
Hello easyw,

I tried the fix, got synthax error :). 
Could you please check it? 

Thank you
CsabaD

[image]

[image]

closed 2025-04-15 by csbflx

csbflx, 2025-04-15:
Updated =>Tried =>Works well