easyw/kicadStepUpMod #252: Loading Board fails after update (ksuToolsOpenBoard)

Babbba13, 2025-04-15:
After an update, the workbench stopped working with this error message:

                      Running the Python command 'ksuToolsOpenBoard' failed:
                      Traceback (most recent call last):
                        File "C:\Users\Andre\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\.\kicadStepUpCMD.py", line 1207, in Activated
                          import kicadStepUptools
                        File "C:\Program Files\FreeCAD 1.0\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
                          return original_import(name, *args, **kwargs)
                                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
                      
                      invalid syntax (kicadStepUptools.py, line 15556)

That is the case on two win11 machines running FC 1.0.0 and KiCad 9.0.1. 

This is the used version:
[image]

This is the error message in FC 1.1.0 dev:
                              Running the Python command 'ksuToolsOpenBoard' failed:
                              Traceback (most recent call last):
                                File "C:\Users\Andre\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\.\kicadStepUpCMD.py", line 1207, in Activated
                                  import kicadStepUptools
                              
                              invalid syntax (kicadStepUptools.py, line 15556)
                              
I searched the internet, FC forums, KiCad forums and found nothing similar.

dimdim123, 2025-04-15:
It's not just loading board. Almost every button do not work at the moment with similar error This is what happened after I tried opening ksuTool:
`invalid syntax (kicadStepUptools.py, line 15556)Running the Python command 'ksuTools' failed:
Traceback (most recent call last):
  File "C:\Users\NB-01\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\.\kicadStepUpCMD.py", line 884, in Activated
    import kicadStepUptools
  File "E:\Program Files\FreeCAD 1.0\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

invalid syntax (kicadStepUptools.py, line 15556)`

Babbba13, 2025-04-15:
I think I found the problem:
there was a line break missing.

[image]

It works now.

I need to find out how to edit that and make a pull request, right? New to this programming thing....

easyw, 2025-04-15:
@Babbba13 
thanks,
I think I have fixed this issue w/ https://github.com/easyw/kicadStepUpMod/commit/a357a1ca31eb7d9994e8401eb1afc3f28a01f3b6
please update the kSU wb

closed 2025-04-17 by easyw

Babbba13, 2025-04-18:
Thank you very much for the fast response and action. It works again.