easyw/kicadStepUpMod #253: Error while loading tracks

simoneamadori, 2025-04-16:
While loading a pcb made with kicad 9.0.1 I get this error:

[image]

and the top tracks are fine, but the bottom are a mess.

Looking in the pcb file the layers are:
```
         (layers
		(0 "F.Cu" signal)
		(2 "B.Cu" signal)
		(13 "F.Paste" user)
		(15 "B.Paste" user)
		(5 "F.SilkS" user "F.Silkscreen")
		(7 "B.SilkS" user "B.Silkscreen")
		(1 "F.Mask" user)
		(3 "B.Mask" user)
		(17 "Dwgs.User" user "User.Drawings")
		(19 "Cmts.User" user "User.Comments")
		(21 "Eco1.User" user "User.Eco1")
		(23 "Eco2.User" user "User.Eco2")
		(25 "Edge.Cuts" user)
		(27 "Margin" user)
		(31 "F.CrtYd" user "F.Courtyard")
		(29 "B.CrtYd" user "B.Courtyard")
		(35 "F.Fab" user)
		(33 "B.Fab" user)
		(39 "User.1" user)
		(41 "User.2" user)
		(43 "User.3" user)
		(45 "User.4" user)
	)
```

while an old file made with KC8 are:

```
(layers
    (0 "F.Cu" signal)
    (31 "B.Cu" signal)
    (32 "B.Adhes" user "B.Adhesive")
    (33 "F.Adhes" user "F.Adhesive")
    (34 "B.Paste" user)
    (35 "F.Paste" user)
    (36 "B.SilkS" user "B.Silkscreen")
    (37 "F.SilkS" user "F.Silkscreen")
    (38 "B.Mask" user)
    (39 "F.Mask" user)
    (40 "Dwgs.User" user "User.Drawings")
    (41 "Cmts.User" user "User.Comments")
    (42 "Eco1.User" user "User.Eco1")
    (43 "Eco2.User" user "User.Eco2")
    (44 "Edge.Cuts" user)
    (45 "Margin" user)
    (46 "B.CrtYd" user "B.Courtyard")
    (47 "F.CrtYd" user "F.Courtyard")
    (48 "B.Fab" user)
    (49 "F.Fab" user)
    (50 "User.1" user)
    (51 "User.2" user)
    (52 "User.3" user)
    (53 "User.4" user)
    (54 "User.5" user)
    (55 "User.6" user)
    (56 "User.7" user)
    (57 "User.8" user)
    (58 "User.9" user)
  )
```

If I overwrite the old layers to the new file it loads without a problem.
I'm doing something wrong with KC settings?

Thanks

easyw, 2025-04-17:
AFAIK till now the name was arbitrary, the number not ...
0 top layer, 31 bottom layer

Do you have a PCB file to share? (Zip it to upload)

simoneamadori, 2025-04-17:
Sure, here an example.

[DC-Load_PiPico.zip](https://github.com/user-attachments/files/19795570/DC-Load_PiPico.zip)

easyw, 2025-04-17:
`  (layers
    (0 "F.Cu" mixed)
    (31 "B.Cu" mixed)
    (32 "B.Adhes" user "B.Adhesive")`
this file seems to be as the standard layer association

simoneamadori, 2025-04-17:
Sorry I uploaded the wrong file...

[DC-Load_PiPico_V2.zip](https://github.com/user-attachments/files/19796651/DC-Load_PiPico_V2.zip)

easyw, 2025-04-17:
it seems they have changed the way they assign layers...
ones it was 0 top and 31 bot, with a whatever name the user could assign.
Now it seems the fixed is the name 'F.Cu' and 'B.Cu', where the number will be assigned depending on the list position...

easyw, 2025-04-17:
It can be fixed, but it will take a bit of time.

easyw, 2025-04-19:
it should be fixed w/ https://github.com/easyw/kicadStepUpMod/commit/026ebcf5b5132057d8b710e1a861e1f71f165d9e
thanks for the testing file :)

closed 2025-04-19 by easyw

simoneamadori, 2025-04-22:
Thank you for your time and effort :)