easyw/kicadStepUpMod #254: CPU hogging?

fakatwork, 2025-04-30:
Don't know if it's due to my particular setup or not, but as soon as I try to load a PCB, FreeCAD maxes out one CPU core and the import takes ages. But the problem is that it stays this way afterwards and never recovers, FreeCAD stays unresponsive, having to wait several seconds after any click (on a rather modern machine)

Setup: Win10/x64
FreeCAD: 1.0.0
WB: 11.6.9

Any test I could perform or extra info required?

fakatwork, 2025-04-30:
After saving the FreeCAD document some errors popped-up in the log:

[image]

easyw, 2025-04-30:
hi @fakatwork 
it seems that before import the board, you have exported some 3D file:
```
Type Notifier Message  
Unhandled std::exception caught in GUIApplication::notify.  
The error message is: bad allocation (2746170 times)  
C:/Users/Alive/Desktop/EE-Mech1.step file amended  
exporting hierarchy  
Unhandled std::exception caught in GUIApplication::notify.  
The error message is: bad allocation (274392 times)  
W:/KiCAD/KZN/EE-Mech1  
Unhandled std::exception caught in GUIApplication::notify.  
The error message is: bad allocation (670472 times)  
select something to be exported!  
Unhandled std::exception caught in GUIApplication::notify.  
The error message is: bad allocation (975168 times)  

```
When there is a `Unhandled std::exception caught in GUIApplication::notify` the program should be closed to avoid other issues.
Anyway you could post here:
1) the file you have exported to "EE-Mech1.step" and the exported one
2) a minimal sample board with the following 3D models ("Holder18650 Full.STEP"  and "Littelfuse_451_NANO2.step")

You can zip and add the files here at GH.

easyw, 2025-05-22:
I cannot replicate the issue. Without a FC sample file no way to test it.

closed 2025-05-22 by easyw