easyw/kicadStepUpMod #258: Unable to install kicad step up on freecad 1.0

amreo, 2025-07-11:
(also see https://github.com/FreeCAD/AddonManager/issues/132#issuecomment-3059331646)
(woodworking has a similar bug: https://github.com/dprojects/Woodworking/issues/60)

I am not able to install KicadStepUpMod.
When I try to install it, freecad shows me this error

```
Git returned a non-zero exit status: -1
Called with: /app/bin/git clone --recurse-submodules https://github.com/easyw/kicadStepUpMod /home/amreo/.var/app/org.freecad.FreeCAD/data/FreeCAD/Mod/kicadStepUpMod

Returned stderr:
Clone in '/home/amreo/.var/app/org.freecad.FreeCAD/data/FreeCAD/Mod/kicadStepUpMod' in corso...
fatal: EOF prematuro
```

But git works correctly if I run the command inside the flatpak.

Step to reproduce:
1) Install freecad from flatpak
2) Go to the addons
3) Search for kicadstepup
4) Install the addon.

Full info
```
OS: KDE Flatpak runtime (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.0.1.39285 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 878f0b8)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.12.10, Qt 6.9.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * FusedFilamentDesign 0.25.250
  * 3DfindIT 1.2.0
  * sheetmetal 0.7.24
  * addFC 1.6.4
  * AddonManager 2025.6.29
  * AnimationFreeCAD 1.0.0
  * Curves 0.6.64
  * Beltrami 1.1.0
  * BillOfMaterials 1.0.3.3
  * boltsfc 2022.11.5
  * Cables 0.2.1
  * CadbaseLibrary 2.0.0
  * cadquery_module 2.1.0
  * CfdOF 1.31.4
  * CurvedShapes 1.0.13
  * DynamicData 2.74.0
  * EasyProfileFrame 0.0.1
  * EM 2.1.1
  * ExplodedAssembly
  * fasteners 0.5.38
  * FeedsAndSpeeds 0.6.0
  * freecad.gears 1.3.0
  * lattice2 1.0.0
  * LCInterlocking
  * MakerWorkbench 1.0.1
  * OpticsWorkbench 1.2.2
  * osh-autodoc-workbench 0.2.3
  * Telemetry 1.0.1
  * kicadStepUpMod 11.7.2
  * woodworking
```

easyw, 2025-07-12:
> The ‘Fatal: early EOF’ error in Git typically occurs during cloning, fetching, or pushing operations. This error is often caused by network issues or repository corruption, leading to an incomplete or interrupted data transfer.

easyw, 2025-07-12:
I checked with latest AppImage https://github.com/FreeCAD/FreeCAD/releases/tag/weekly-2025.07.07
and I can reproduce the issue.
But if I call the git command with a terminal command, everything goes fine.
`/usr/bin/git clone --recurse-submodules https://github.com/easyw/kicadStepUpMod /home/user/.local/share/FreeCAD/Mod/kicadStepUpMod`
`Cloning into '/home/user/.local/share/FreeCAD/Mod/kicadStepUpMod'...`
`remote: Enumerating objects: 3793, done.`
`remote: Counting objects: 100% (1003/1003), done.`
`remote: Compressing objects: 100% (126/126), done.`
`remote: Total 3793 (delta 932), reused 898 (delta 877), pack-reused 2790 (from 2)`
`Receiving objects: 100% (3793/3793), 125.71 MiB | 2.00 MiB/s, done.`
`Resolving deltas: 100% (2780/2780), done.`
It seems an issue in the AddonManager.

at stackoverflow you get:
https://stackoverflow.com/questions/21277806/fatal-early-eof-fatal-index-pack-failed
in an answer there is a tip:

> This error may occur for memory needs of git. 

Hope this may help in reporting the issue to FreeCAD Addon Manager.